tatobari/hx711py #1: Inconstant value returned

eddie4, 2016-06-06:
First off let me say thanks for the code. However am still getting inconstant values.

A typical measurement looks as follows:
-30217
-30187
2742472
2742470
-30228
-30231
645318
-30186
2742424
-20956
-29201
-30305
-30259

When I use the exact same setup with https://github.com/ggurov/hx711 which is written in C. I don't have the issue. However it's C and is going to be a pain to get that into flask. Any advice/solution?

tatobari, 2016-06-06:
Hi Eddie,
A few questions to know where we're standing.

1. Can you send a pic of the HX711 you're using? Is it mounted in one of
those green or red boards or did you just get the stand alone chip?

2. Did you change any of the values I use on the "configuration methods" in
the example at the end? I use the "set_scale" to simply remove digits, by
division, from the very large number returned by the hx711. The
"set_one_kilo" is the value of one kilo after removing the digits. I
"tared", checked what was the reading with 1kg and then entered the value.

We can have a skype or hangout if you want. Let me know.

Tato

tatobari, 2016-06-06:
I've just got home and tested everything again to make sure everything is ok. I even made a small modification to the code just to power off the chip at the beginning in case I shut the program and let the HX711 in on in the middle of a transfer.

I've tested (using a very expensive multimeter that a friend lend me) the 5v and the 3.3v power pins on the Raspberry Pi 2 Model B I've got here. 3.3v is WAY more stable than 5v. On my Arduino UNO, both are pretty stable. I'd use the 3.3v of the Pi in case you were using the 5v.

These are my readings right now. Check how 3.3v is much stable.

**No weight 5v (1.0 = 1kg) on a 20kg LoadCell:**
0.000
0.728
0.000
0.728
0.000
0.000
0.000
0.000
0.000
0.000
0.739
0.739
0.000
0.000
0.761
0.000
0.000
0.000
0.739

**No weight 3.3v (1.0 = 1kg) on a 20kg LoadCell:**
0.000
-0.011
-0.011
-0.011
-0.011
0.000
-0.011
0.000
0.000
0.000
-0.011
0.000
0.000
0.000
-0.011
0.000
0.000
0.000
0.000
0.000
0.000

**Note:** Make sure not to leave your phone close to the hx711 or the load cell cables. The phone waves can really mess up the readings. I've waisted about 2 hours this weekend trying to figure out why was I getting peaks of 45kg on a 20kg loadcell with nothing on it.

eddie4, 2016-06-08:
Ill check when i get home. Am pretty sure i have it connected to the 5V instead of 3.3v. ill make a change there. 

I have the following HX711:
http://www.ebay.com/itm/281925028853?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

tatobari, 2016-06-08:
We've got the same HX711 board. Don't forget to pull the latest changes to
the code.

By the way, my code is using GPIO5 (Pin 29) for the DT Pin on the HX711 and
GPIO6 (Pin 31) for the SCK Pin on the HX711. There params are set when you
instantiate the class.

Looking forward to get good news from you.

tatobari, 2016-06-12:
I haven't received any answers from you in a while. I'll consider this closed. There have been some updates to the code which might help you.

eddie4, 2016-06-14:
Sorry for the late response. The issue still exists (I got the newest code) but I believe that it has to do that am using a RP1 first edition. Which is not overclocked and perhaps when slightly busy screw's up the result. Am going to see if ADS1115 Module is a good replacement for the HX711. Ill report back but as it's coming from china it's going to be a while.

tatobari, 2016-06-14:
Hi,
I'm not sure if ADS1115 amplifies the signal enough. I will reopen the
ticket. I really suggest that you read the other open issue that's in this
repo. The other guy has the exact same problem and we've walked through it
step by step.

Raspberry 1 should work anyway. If you see any delays on the printing of
the code it is mostly due to the hx711 not being "ready" to transmit. This
may be due to an incorrect wiree bridge. I don't mean to underestimate you
but if you could tell me what kidnd of load cell or load cells you have
(how many, weight, three or four wires) I might be able to help you, at
least hile you wait of the ADS.

Let me know and I'll re open the issue so we can solve it.

closed 2016-06-20 by tatobari

tatobari, 2016-09-09:
Hi, there was a problem with the code regarding the order of the bits. I'll be updating the code so the order of the bits can be set up by configuration.