tatobari/hx711py #2: How to tare & define 1kg

u-l-m-i, 2016-06-10:
ello Tato,

thanks for providing your code and explanation,
but I’m still struggling with a few things.

First: Can you please explain how to evaluate the tare value and the value for the defined (1kg) 

Second: Maybe depended on above

With changing the Vcc to 3,3 V the results get better, but still a very inconstant.
My scale consists out of 4 Load Cells which are linked as Wheatstone Bridge.
(tare & 1kg as in our example)

e.g.
Empty scale:
pi@raspberrypi:~/scale/hx711 $ python hx.py 
0.000
1.076
-0.043
4.500
-0.033
0.000
-0.033
0.000
0.022
-0.022
-0.033
0.022
-0.011
-0.011
-0.022
^CCleaning...
Bye!
pi@raspberrypi:~/scale/hx711 $ python hx.py ## with 1kg
0.000
0.043
0.043
0.033
18.261
0.011
0.196
0.022
0.087
0.087
0.000
0.011
^CCleaning...
Bye!
pi@raspberrypi:~/scale/hx711 $ python hx.py ##wtih 5kg

0.000
0.022
0.033
0.033
0.076
0.043
0.043
0.065
-0.011
0.022
0.033
^CCleaning...
Bye!
pi@raspberrypi:~/scale/hx711 $ 

Any idea or advice?

Thanks 
u-l-m-i

tatobari, 2016-06-10:
Hey u-l-m-i!
I'm sorry that the code is not that tidy yet. Down to business.

*Code Updated*
I changed the "one_kilo" vars and methods to "reference_unit". Makes more
sense.

*How I "fixed" the code*
To figure out what was wrong, I decided to print the 8bit arrays the code
extracts from the HX711 when reading. I found out that they were being
interpreted as LSB Bytes with MSB Bits while they where MSB Bytes with MSB
Bits.

Remember? MSB means "Most Significant Bit First" and LSB means "Less
Signifficant Bit First".

It's was little odd (it's wierd how you can't use the word "bit" to say
"little" on this kind of topics), because most ICs use the LSB rule. Once I
changed that part of the code to switch the bytes before building the
integer and build up an integer from the three bytes in the correct order,
without any load cell wired to the HX711, I began to get readings
of 8385000 +/- 100. I new I was on the right track. It wasn't jumping from
-4 million to +16 millon. It was making much more sense.

*Getting a reference unit*
I took a leap of faith and wired the load cell and run the code again. The
value changed to 8160500 +/- 100. It was still pretty stable. After that, I
added 2kg, 2L plastic bottle filled with water (water density is 1). The
reading changed to 8348600 +/100 and was still very stable. So I assumed
that 8160500 - 8348600 = 188100 should be about 2kg, hence, 94050 should be
1kg.

If you think about it, 0,94kg are 940g which are 94000cg (centigrams). It
depends on the loadcell obviously, but this was a great clue that I was on
the right track.

In the code I've set the reference unit to 92. I'm doing all this process
again while I write this and I'm using a different loadcell now.

*Tare*
hx.tare() gets an average of 5 readings, sets the "OFFSET" variable and
will substract that variable from every raw reading. The "scale" and the
"reference_unit" values are all used after using the "OFFSET" variable.

*How I meassure*
"set_scale(int)" will divide every raw value by the in specified in the
argument.

"set_reference_unit(in)" will also divide the value by the integer
specified in the argument keeping three decimals to the result of the
division.

*How you should meassure*
Use the get_value() method. It will give you a raw value so you can check
what is the relation between the returned values, your load cell and the
weight you are using. The example code would be:

############# EXAMPLE
hx = HX711(5, 6)
hx.power_down()
hx.power_up()
hx.tare()

while True:
    try:
        val = hx.get_value()
        print val

        hx.power_down()
        hx.power_up()
        time.sleep(0.5)
    except (KeyboardInterrupt, SystemExit):
        cleanAndExit()

Hope it helps.

u-l-m-i, 2016-06-10:
Thank you for your comprehensive answer -

First I'll check the returned values of my hardware setup without the connected load cell to see if this working with your description to get the values.

tatobari, 2016-06-10:
Hi,
You may most probably not get the exact same values I am getting. I guess
they depend a lot on your electric company and other stuff like that, but I
guess it shouldn't vary to much.

Do not hesitate to contact me if you have any more doubts.

Tato

u-l-m-i, 2016-06-10:
Hi Tato,

without connected load cell - only HX711 - I'll get still no reasonable stable numbers with the code example for val = hx.get_value().

pi@raspberrypi:~/scale/hx711 $ python hx.py
271
34363
-317
33924
26
635
33528
-140
155
-453
-616
-416
-567
-287
-521
-307
-593
33816
-134
438
-401
755
^CCleaning...
Bye!

tatobari, 2016-06-10:
Are you running any of these methods?
- ```hx.set_scale([any integer])```
- ```hx.set_reference_unit([any integer])```
- ```hx.tare()```

If you're are only running ```hx.tare()```, then you're on the right track and I'll explain as soon as you confirm.

u-l-m-i, 2016-06-11:
I'm very curious!

Code is currently:

hx = HX711(23, 24)
hx.power_down()
hx.power_up()
hx.tare()

while True:
    try:
        val = hx.get_value()
        print val

        hx.power_down()
        hx.power_up()
        time.sleep(0.5)
    except (KeyboardInterrupt, SystemExit):
        cleanAndExit()

tatobari, 2016-06-11:
Hi,
Then you're on the right track. These values you're reading are very small
variances. Try wiring you're loadcell and add and remove weight that is, at
least, 10% of what your load cell is for. Send me the values and I'll
explain.

u-l-m-i, 2016-06-11:
Hi,

load cell rewired 

Values with empty load cell vs load cell with 75kg

8384796		8379370
8390597		8390759
8383937		8385978
8391525		8387135
8386185		8390192
8378267		8383300
8384679		8387200
8382311		8389063
8386792		8389179
8384580		8391757
8388173		8384454
8383990		8386172
8391985		8383341
8389005		8382682
8388842		8390789
8396705		8388998
8385658		8382688
8387546		8378510
8395684		8389982
8382082		8378329
8388465		8390651
8389040		8386372
8379458		8387811
8388029		8733460

tatobari, 2016-06-11:
So, you're getting values around 838000 +/- 1000. Your differnces are above
min by the order of 10. This could probably be due to a soldering problem.
I'd check that. However, you're receiving very good values for a 75kg load
cell.

I'll get back to you tomorrow with more details, I'm not at the computer
right now.

u-l-m-i, 2016-06-11:
Just to keep you updated:

I re-soldered all pins but still have this huge differences.
The load cell itself is from an outdated body scale -

Giving up (at least for today)

tatobari, 2016-06-11:
Ok. What I was trying to tell you is that the differences you're getting
correspond to less than 1kg differences. Those are not huge differences in
a 75kg loadcell. It's actually pretty good.

If you can make another list of values that correspond to something like
5kg, I'll send you the code so that you can get proper values in kg.

u-l-m-i, 2016-06-11:
Here are the new values:

0kg		         5kg
8375532	|	8377626
8379648	|	8373439
8374981	|	8381200
8380666	|	8382726
8378561	|	8379245
8376800	|	8377681
8370368	|	8372074
8387048	|	8381275
8373232	|	8374777
8374336	|	8370207
8381691	|	8375596
8381293	|	8375067
8380966	|	8376609
8380530	|	8375279
8368901	|	8380899
8377193	|	8370332
8373520	|	8370963
8375517	|	8374112
8376315	|	8375616
8374361	|	8375368
8382907	|	8378521
8380854	|	8377200
8373578	|	8375083
8378806	|	8381274
8372359	|	8377061
8378776	|	8378717
8381503	|	8379099
8385736	|	8380025
8372550	|	8379171
8376316	|	8371412
8373451	|	8374168
8376962	|	8377284

tatobari, 2016-06-12:
(Sorry for the delay, I'm away for the weekend and I don't have a good connection.)

So, looks like you're readings are pretty much the same with and without weight, which is a problem. It might sound like if I were underestimating you but it's better to ask and be sure than to assume. Here it goes:

**1- HX711 Noise**
On a previous post, the values of your HX711 without anything wired, flicker by not more than 500 "points" except for a few jumps. That's very little noise and it's ok.

With the load cells wired, the flickers are over 5000 "points". There seems to be a lot of noise and, since you already resoldered the HX711, I'm gonna ask you about the load cell and the bridge, on point 2.

**2- Load Cells and Wheatstone Bridge**
How many cables do each load cell have? It should be either three or four cables. Can you send a pic of the bridge and the HX711 so I can see how are they wired?

Thanks! Don't loose your motivation, we're getting there.

u-l-m-i, 2016-06-13:
Hi Tato,

after several tests I also suspect the load cell as root cause.
I'll check the 4 load cells (3 wires each) & Wheatstone Bridge during this week and will update as soon as it's possible.

Thanks for your support & your patience  !

tatobari, 2016-06-13:
Hi!
I'm not sure if you are aware but three wire load cells are basically
half-wheatstone bridges and wiring it is not as intuitive as it looks.

I've found a very usefull product in sparkfun.com that may help you with
that. Here's the link:

https://www.sparkfun.com/products/13281

However, if, like me, you're not in a position where getting those products
is easy (I live in Argentina), here is the link to the schematics of the
combinator so you can wire the load cells yourself pretty easily:

https://cdn.sparkfun.com/datasheets/Sensors/ForceFlex/SparkFun%20Load%20Sensor%20Combinator%20v10.pdf

Just ignore the RJ45 and JP7 parts. In those schematics, JP5 is the HX711
with 5 pins instead of 6. Just ignore the 5th pin (the shield pin) and wire
the bridge you made according to the schematics to the other 4 pins just
like the schematic indicates.

Let me knw if you have any doubts.

Tato

u-l-m-i, 2016-06-19:
Hello Tato,

still encountering problems and hope on your support.

As I didn’t trusted my soldering on the wheatstone bridge I ordered the spark fun wheatstone bridge and soldered it yesterday.

And the results changed - but not to the positive.

Without having the load cell attached to the HX711 I get the following result, which looks quite to what I was used to.

8387104
8385288
8389014
8385519
8387639
8385746
8387182
8385996
8387657
8386316
8387636

With load cell attached to the HX711 the results changed:

16777215
16777215
16777215
16777215
16777215
16777215
16777215
16777215
16777215
16777215

Those values looks like the the double value of above and also to theoretical max value.

It doesn’t make a difference if the A+ / A- are connected.
Weight on the load cell also doesn’t affect the results.

The resistance are the following

E+ / E-	1050 Ohm
A+ / A-	1050 Ohm
E- / A-  700 Ohm
E- / A+  700 Ohm
E+ / A- 1050 Ohm 
E- / A+ 1050 Ohm

As the program works like a charm without the load cell I assume the HW setting inkl. the HX711 fine (as I get reasonable results with your code).

So I’m asking if you have any idea how to determine the root cause of my problem?

Thanks in advance

tatobari, 2016-06-20:
This looks really strange to me. I hope I'm not killing your motivation but this really looks like, either a wiring, soldering or hardware issue Do you think you can attach a few pics of the combinator, the load cells and the HX711? It would make it much easier for me to debug the problem.

I'm pretty sure there's something wrong here because the theoretical max value repeating over and over means there is a short circuit. Be careful with this, you could end up blowing the HX711 IC.

On the other hand, your "idle" values flickr too much. I wonder if it's a problem with the electrical service where you are or something is not quite properly wired.

u-l-m-i, 2016-06-21:
Hi,

as I took a blown HX711 into account I ordered already a new HX711 - but it will take some time to get it shipped from china.

If possible I'll take a picture of the spark fun combinator and the connection to the raspi.
The load cell wiring it self is not easy to picture as it is a old body scale - with wires extended form inside out.

As I just tried to take the pictures I determined a broken wire, so have to correct the wiring and update.

Thanks for staying with me!

u-l-m-i, 2016-06-21:
[image]
[image]

To fix the loose pin didn't change the result.

arthurmoises, 2016-06-24:
Hi tatobari! Thanks for the great code and great explanations! 

I'm having some issues, and I guess you can help me. 

I'm using a green hx711, 4 load cells fixed into a ~50cmx50cm wood board, with rubber supports between the sensor and the ground. The hx711 is connected trough raspberry PI2 at pins 08(GPIO14) and 10 (GPIO15). 

I'm running the following code: 
'''
hx = HX711(14, 15)
hx.power_down()
hx.power_up()
hx.tare()

while True:
    try:
        val = hx.get_weight()
        print (val)

        hx.power_down()
        hx.power_up()
        time.sleep(0.5)
    except (KeyboardInterrupt, SystemExit):
        cleanAndExit()
'''

And I'm getting the following values:
4228.000
4368.000
4196.000
4368.000
4196.000
4368.000
4026.000
4368.000
4282.000
4368.000
-3825.000
4368.000
4282.000
4368.000
4111.000
4368.000
-17478.000
4368.000
4282.000
4368.000
4282.000

Can I assume that the 0kg load (tare) is 4368.000?

What are the next steps to convert to kg?

Thanks in advance!

tatobari, 2016-06-24:
Hi u-l-m-i and Arthur,
I'm out of town until Wednesday. Been out of town since last Friday. I'm
really sorry for not letting you know before, u-l-m-i.

I'll try to give you both an answer tomorrow since I finally got to a place
with a fairly good internet connection.

Please hold on!

Tato

arthurmoises, 2016-06-26:
Hi Tatobari and u-l-m-i!

I've managed to get the load cells working on Friday! Finally! Now it's just a matter of configuring Tkinter correctly to finish my project. The code runs very well, thanks a lot for the port and fixes, Tatobari.

All the incorrect readings were because of a wrong level of one of the load cells. It wasn't touching ground. A friend of mine realized that and fixed. After that... ta-da! 81,9kg in the monitor. 

The steps that I took were: 

1- Run the script as-is: (get_value and hx.tare()) If it's stable, move on! If not, check wirings of load cells.
2 - Run it again with get_weight() Take note of average readings on a paper
3 - Run it again with a known weight (I steeped into it with 81,9kg) Take not of that average value too.
4 - Subtract the two previous values and divide by weight (I got 120, aproximatedly(117))
5 - Now uncomment scale and reference_unit, which I set to 120 as a argumment (reference_unit(120)) 
6 - Now, here lies something that I don't know yet: I was getting values/10 of the real ones (8,2kg, instead of 82kg)
7 - To correct that, I've set scale to (100) instead of scale(1000). It worked, I was getting aproximatedly values of 81,9 kg!

Thanks a lot, Tatobari and u-l-m-i!

u-l-m-i, 2016-07-19:
Hi Tatobari,

just to let you know: 
my load cell was faulty - the new HX711 had the same issue, so I replaced the load cell and it worked. 
Still having amplitude variations but in summary it works and I get reasonable and reproducible results.

Thank you for your  code, your help  and assitance !

closed 2016-07-19 by u-l-m-i

tatobari, 2016-07-19:
Well those are great news. I'm really sorry I couldn't give any more
support. The past few weeks have been a bit rough.

faurepa, 2017-04-02:
Great comments, examples and discussion.  I thas helped me figure out how to use this code.

thank you

SooranSaeed, 2018-08-11:
hi tatobari im using chinese hx711 thats show in this pic below
im using your library and work fine but my max value in average reading is 16777215
16777215
16777215
8388607
8388607
but when i put nothing on loadcell is 8388607 and when i put 1kg weight on loadcell is 16777215
do you think my trouble is for my hx711 module ?

[image]
[image]

tatobari, 2018-08-13:
@SooranSaeed Hey man! The bad news is that you probably borke the little board. The soldering looks all overheated and the pads broken.

The A-, A, Out+ and GND pins are extremely sensitive and they need almost perfect soldering. You're look really really bad, I'm sorry to tell you. This one was soldered by me:

[image]

[image]

Let me tell you, even my soldering is not good at all. Try going into YouTube on "how to Solder pins", such as [this one](https://www.youtube.com/watch?v=3230nCz3XQA).

On the other hand, I've never seen that HX711 implementation circuit before. I always use the following type of implementation:

[image]

More detail [here](https://www.mouser.com/ds/2/813/hx711_english-1022875.pdf).

SooranSaeed, 2018-08-14:
hey tatobarii im buy new hx711 that show below 

my get_weight(5) value without set refrence unit is 

0
0
0
0
6710886
8388608
8388608
8388608
8388608
8388608
8388608
8388608
6710886
8388608
6710886
8388608
8388608
6710886
6710886
3355443
0

my load cell is 50 kg and put 1kg on scale just show 8388608 and when i increase weight and put 3 4 or 5 kg on scale just show 8388608 and when i set refrenceunit(92)

show 

0
0
0
91180
91180
91180
60787
91180
91180
91180
60787
91180
60787
91180
60787
60787
91180
60787
60787
0

and my max value is always 91180

is your library true ?? this is freaking me out 

[image]
[image]

tatobari, 2018-08-14:
@SooranSaeed 

1. Could you send a picture of the loadcell? The series of numbers looks really odd. I wonder if some wiring is wrong.

2. The HX711 looks great, by the way.

3. What pins are you using on the Raspberry?

4. Can you share some of the code?

SooranSaeed, 2018-08-14:
im using beagle bone black nor rasberrypy 
im using GPIO pin in beaglebone black to set CLK pin HIGH and LOW
and Data pin is GPIO pin Which i read their 1 or 0 

this is my __init__.py
----------------------------------------
`import Adafruit_BBIO.GPIO as GPIO
import time
import numpy  # sudo apt-get python-numpy

class HX711:
    def __init__(self, dout, pd_sck, gain=128):
        self.PD_SCK = pd_sck
        self.DOUT = dout

        GPIO.setup(self.PD_SCK, GPIO.OUT)
        GPIO.setup(self.DOUT, GPIO.IN)

        self.GAIN = 0
        self.REFERENCE_UNIT = 1  # The value returned by the hx711 that corresponds to your reference unit AFTER dividing by the SCALE.

        self.OFFSET = 1
        self.lastVal = long(0)

        self.LSByte = [2, -1, -1]
        self.MSByte = [0, 3, 1]

        self.MSBit = [0, 8, 1]
        self.LSBit = [7, -1, -1]

        self.byte_format = 'LSB'
        self.bit_format = 'MSB'

        self.byte_range_values = self.LSByte
        self.bit_range_values = self.MSBit

        self.set_gain(gain)

        time.sleep(.1)

    def is_ready(self):
        return GPIO.input(self.DOUT) == 0

    def set_gain(self, gain):
        if gain is 128:
            self.GAIN = 1
        elif gain is 64:
            self.GAIN = 3
        elif gain is 32:
            self.GAIN = 2

        GPIO.output(self.PD_SCK, GPIO.LOW)
        self.read()

    def createBoolList(self, size=8):
        ret = []
        for i in range(size):
            ret.append(False)
        return ret

    def read(self):
        while not self.is_ready():
            # print("WAITING")
            pass

        dataBits = [self.createBoolList(), self.createBoolList(), self.createBoolList()]
        dataBytes = [0x0] * 4

        for j in range(self.byte_range_values[0], self.byte_range_values[1], self.byte_range_values[2]):
            for i in range(self.bit_range_values[0], self.bit_range_values[1], self.bit_range_values[2]):
[... 137 lines cut ...]
# In this case, 92 is 1 gram because, with 1 as a reference unit I got numbers near 0 without any weight
# and I got numbers around 184000 when I added 2kg. So, according to the rule of thirds:
# If 2000 grams is 184000 then 1000 grams is 184000 / 2000 = 92.
# hx.set_reference_unit(113)
#y
hx.set_reference_unit(8388)
hx.reset()
hx.tare()

while True:
    try:

        val = hx.get_weight(5)
        print(val)

        hx.power_down()
        hx.power_up()
        sleep(0.1)
    except (KeyboardInterrupt, SystemExit):
      cleanAndExit()

`

i think my truoble is my beaglebone black but i used  GPIO pin as same as rasberypi
for end in my init.py file i just remove GPIO.setmode(GPIO.BCM) cause in beaglebone black we havent this line please help me tatobari im realy need this loadcell

tatobari, 2018-08-14:
@SooranSaeed Could you send a picture of the end of this cable? Thanks!

[image]

SooranSaeed, 2018-08-14:
my brother tatobari im soldering my loadcell fine look like that : A+(white) , A-(Green),E-(Black),E+(Red)
my library is fine but my big truoble is my weigh scal not going up , i was calculate my refrence unit and that is 8388 and when i put it get_weight methos print 1000 and that is 1 kg but when i put 2 kg or 4 or 6 kg my value is stck and show just 1000 and not going up , my truoble is that

[image]

SooranSaeed, 2018-08-14:
Tatobari do you have whatsapp or facetime or ino or anything to call with you i really need your help , please email me your id or number this is my email : saeed.nazari92@gmail.com
If you have instagram you can follow my id : 
Sooranpager

tatobari, 2018-08-14:
@SooranSaeed Could you improve the BeagleBone picture? I'd like to see that closer. Thanks!

SooranSaeed, 2018-08-14:
[image]

this is my beaglebone black i think your code not fine with my beagle bone , could i write this library for myself ?   please help me

tatobari, 2018-08-14:
Please, do the following and send me the results:

```python
import Adafruit_BBIO.GPIO as GPIO
from time import sleep
import sys
from hx711 import HX711

def cleanAndExit():
    print "Cleaning..."
    GPIO.cleanup()
    print "Bye!"
    sys.exit()

hx = HX711("P9_12", "P9_16")

hx.set_reading_format("LSB", "MSB")

#hx.set_reference_unit(8388)
hx.reset()
hx.tare()

while True:
    try:

        #val = hx.get_weight(5)
        val = hx.get_binary_string()
        print(val)

        hx.power_down()
        hx.power_up()
        sleep(0.1)
    except (KeyboardInterrupt, SystemExit):
      cleanAndExit()
```

SooranSaeed, 2018-08-14:
for first time i put nothing on scale , and then i put 1 kg on scale and in line 14 you see value changes

0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b01111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000 
0b11111111 0b11111111 0b11111111 0b00000000

tatobari, 2018-08-14:
Looks like the voltage is permanently high, except for some times. Try using `VDD 3.3 V` instead of `VDD 5V`. For what I can see [here](https://www.mathworks.com/help/supportpkg/beagleboneio/ug/the-beaglebone-black-gpio-pins.html), the logic is 3V3, not 5V.

On some embedded devices, this doesn't make much of a difference, but it's worth trying in this case. Please, share the results once you have them.

tatobari, 2018-08-14:
Regarding this:

> i think 3.3 v is to high do you want to put this voltage to 1.8 v ? but load cell is work with 3 or 5 volt

You have to power the HX711 with the logic level that the BeagleBone Black GPIO Pins work at. According to the link I've shared before, that value is `3.3 V`. So, connect `Physical Pin 3 (VDD 3.3 V)` to `Vcc Pin` on the HX711.

I didn't understand what you meant by _"pin out the vcc pin"_ here:

> hey bro i put out the vcc pin in beagle bone and this is my result

SooranSaeed, 2018-08-14:
beagle bone GPIO Voltage for normal is 3 v and when i connect 5 v  voltage is raise to 5 volt , i think thos gpio can push the hx711 on , i mean im not connecting vcc of hx711 on beaglebone black

tatobari, 2018-08-14:
> beagle bone GPIO Voltage for normal is 3 v and when i connect 5 v voltage is raise to 5 volt , i think thos gpio can push the hx711 on , i mean im not connecting vcc of hx711 on beaglebone black

There would be power running through the HX711 and you might get some random values because the logic is no the BeagleBone, but the HX711 would not be working properly. I'm honestly running out of ideas.

SooranSaeed, 2018-08-14:
hey bro i measure DT voltage with ohm meter that was 3.3 v  , and my ck votage was 57 mvolt what can i do bro

tatobari, 2018-08-14:
@SooranSaeed I'm sorry, I think I can't help you from here. There seems to be en electrical issue going on there. The library, as far as I've tested it here, works well on Raspberry Pi. Can't really help you. Sorry!

SooranSaeed, 2018-08-14:
Hey it isn’t any way to work with loadcell , can i use I2C port to use loadcell , im so confused

tatobari, 2018-08-14:
The HX711 uses a custom protocol. The library on this repository has been thoroughly tested and works fine. I even tested it right now. You definitely have a wiring issue on your side, or the BeagleBone Black can't handle the protocol as it is developed in the library. I can't support BeagleBone at the moment if that is the problem.

SooranSaeed, 2018-08-14:
When i write that protocol by myself , even though that was not working , so is there any chance ?!

SooranSaeed, 2018-08-14:
My boss said me beaglebone is very powerful than rasberypi and it most be working , whats your idea is it true ?!

tatobari, 2018-08-14:
It is possible that the BeagleBone is too fast and the frequency it's clocking the SCK pin is too high. Try reading the datasheet. If possible, try finding an oscilloscope to find out the frequency. They you can delay the clocking a bit adding some some microseconds delay between line 68 and 69 to respect the minimum microseconds explained on the HX711 datasheet.

tatobari, 2018-08-15:
If you're using a voltmeter, you'll only get an average voltage of a signal that's changing with time. 

```python
GPIO.output(self.PD_SCK, GPIO.HIGH) # Voltage is 3.3V
GPIO.output(self.PD_SCK, GPIO.HIGH) # Voltage is 0.0V
```

But the above happens so fast that your testet won't change fast enough.

If you finally got an oscilloscope, then send me a picture of the screen to see the wave form. Thanks!