tatobari/hx711py #3: resolution?

lonelysushi, 2016-06-15:
Hello,

I am fairly new at this and this HX711 code has worked best for me so far.  However, is there a place that i can change the resolution? for some reason my output is only in integers, it doesnt show decimals in any amount of places (looking for atleast 2 decimal places).

Thanks

tatobari, 2016-06-15:
Hey,
Are you running the code just as it is? In that case, it should be printing three decimals. Let me explain.

The HX711 returns a 24 bit values everytime it is asked for the current weight. Since it is an ADC Amplifier (Analgo.to Digital Amplifier) what it actually does is change Volts to Bits. As you may know, you can make binary numbers (numbers expressed using only to digits, which are 0 and 1) from a bunch of bits you can make 10 digits based numbers (0, 1, 2, 3, 4, 5, 6, 7, 8 and 9).

So, if you convert the 24 bits (which are 24 digits which can be either 1 or 0) to a 10 digits based number, you get and integer.

In the case of my HX711 little green boards, if you don't wire any load cells and just read the hx711 by it's self, you get a value around 8385000, and it drifts from 8384700 to 8385300. Thats is a clean and raw value on which the first three digits (from the right), the ones that drift, correspond to VERY tiny fluctuations on the electricity that goes through the loadcell. Which means that we can just ignore them.

That's when you set the SCALE value. The raw value is divided by the SCALE value (1000 in the example) to remove the digits we don't need.

The REFERENCE_UNIT value is used to change the reading to a proper weight.

I'd recommend you to print hx.get_value() method and add or remove weight. Then use a little math to find out how to convert that number to your reference scale (kilos, grams, pounds, etc).

Let me.know if you meed any more help.

lonelysushi, 2016-06-15:
thanks for the reply, let me try it and get back to you on this.

currently when i run it, it is showing decimal, BUT it is constantly on "xxx.000" which essentially is only showing integers which is why i was confused.  I currently have a separate code to get_value (also tried get unit and get weight just to see what they are) and just print out the result

tatobari, 2016-06-15:
Great. Try to paste here some values (not less than 10) and add the code
you used to get them. Also, specify what kind of load cell you're using
(number of wires), what's its weight scale (5kg? 10kg? ...), and, if
possible, attach a picture of how it is wired to the HX711.

The more info I get, the more I can help you.

lonelysushi, 2016-06-15:
i am currently running a 4 wire 150kg loadcell. each set connected to junction boxes to add them (hoping for a total of 300kg with two loadcells on one platform).  I am also using the green HX711.

do you think its possible to write to you via email at mingroung@gmail.com for further discussion? (not that i theres anything to hide, but am very new at this whole thing and semi embarrassed / dont wanna flood the forum with too tedious of a problem :))

tatobari, 2016-06-16:
A quick update was made to the code to address this issue.

closed 2016-06-16 by tatobari

tatobari, 2016-09-09:
Hi, there was a problem with the code regarding the order of the bits. I'll be updating the code so the order of the bits can be set up by configuration.