tatobari/hx711py #4: Help with Troubleshooting Issue

alotred, 2016-09-06:
Hi Tato

Thank you so much for offering to help. I am using Pi 2 Model B. I have connected the Hx711 as follows:
VCC - 3v3
GND - Ground
CLK - GPIO23
DATA - GPIO24

[image]

I've only changed line 140 in the code to: hx = HX711(23, 24)
After running it with "sudo python hx711.py" I get the following output

[image]

It appears to be very erratic. I have also tried to switch the CLK and DATA pins around and I get the value "18.23" constantly even with the addition of a weight. Thanks again

Regards

tatobari, 2016-09-06:
Hi,
I think you're really close to get it working. Could you also attach a
picture of your loadcell?

Thanks!
Tato

alotred, 2016-09-06:
Hi tatobari,

Here is an image of the load cell.
[image]

Thank you so much for the help!!

Regards

tatobari, 2016-09-06:
Hi,
From the mechanical point of view, one little detail that I'm sure you
might know is that the load cell is suposed to "rest" only on of it sides
while the other side is suspended in the air and holds the weight. The load
cell is suposed to fold when preassure is applied. The picture on this link
should illustrate what I mean.

From a more electronic point of view, I'm gonna need a picture of the other
side of the hx711.

I have to leave to work know but I'll take a look while at the office and
get back to you.

We're almost there. Don't give up.

Tato

alotred, 2016-09-06:
Hi Tato,

In terms of the mechanical side of things, I have discussed this with numerous mechanical engineers and the image i sent you was not how the load cell will be mounted as I do not have that available to me right now.

I have attached the other side of the hx711, I hope this is what you meant :dancer: :) 
[image]

I really appreciate the help!!

Regards

tatobari, 2016-09-06:
Hi,
To clarify, if you set `hx = HX711(23, 24)` then the DATA PIN should be on GPIO23 and the CLOCK PIN should be in PIN GPIO24. That's one thing.

On the other hand, my guess is that you're probably using a protoboard to connect the loadcells to the HX711. The potential differences coming from the loadcell are very very small which make the signal very susceptible to noise. It is good practice to plug the loadcell directly into the HX711 without the protoboard in the middle.

Finally, all that noise you're getting could be due to the `hx.set_reference_unit(92)` and the `hx.set_scale(1000)` parts of the code. Just comment those lines and send me the output so we can have more resolution and, therefore, check how the 24 bits returned from the HX711 are behaving.

Tato

alotred, 2016-09-07:
Hi Tato,

I have checked the pins and they are set just as you have specified and I am not using a protoboard but instead I have directly connected the load cell to the hx711.

I have done as you have asked and the output is the following:
[image]

Sorry for the late reply, our time zones are very different and so I was asleep. Thank you so much for all the help, I can't thank you enough!

Regards

tatobari, 2016-09-07:
Hi,
It's 23hs here in Buenos Aires. Going to bed in a few minutes.

Your log looks good actually. There might be an issue on my side. I think my code is overflowing the "int" value which causes positive or negative numbers where ther should be the oposite.

I'll make a quick change but I won't test it right now cause I don't have my Py here and, besides, I'm really tired. I got back fron the office at 10. Long day.

Stay tuned. I'll answer back in a few minutes.

Tato

tatobari, 2016-09-07:
Hi,
Just did a few modifications to the code. It's not tested though. I think there might be a bit more to do but I guess this could stop the flicks. Let's see what happens. I'll try to get the Py back sometime during this week. It's at my partners house.

Tato

alotred, 2016-09-07:
Hi Tato,

If you upload it I could just test it now for you :) 
Thank you so much for this!

Kind Regards

tatobari, 2016-09-07:
I've already pushed the changes so give it a try! Good luck. I'm sure something will break down. I invite you to try and make changes and corrections to the code, it's really simple. I just didn't one day in a hurry, taking Michael Vine's NodeJS code as a reference to port from.

alotred, 2016-09-07:
Thank you for the quick response. I have been playing around and I've found that the results are still all over the place. I've been trying to understand the code (I have never worked with python before) but I found the results are slightly better when I set the gain to 32 instead of 128. 

[image]

Everytime I run the code. I get very different numbers however, now they are always positive and I get 0.000 often. Because of how erratic the results are, I can't tell if the system can sense a load and so I'm wondering whether the problem is load cell itself.

Kind Regards

tatobari, 2016-09-07:
Hi,
Are you over a deadline? If not, I can get my Raspberry by the weekend,
check if it works and get back to you with more information.

Tato

alotred, 2016-09-07:
Hi,
Yeah, we were meant to have everything running by this Friday which is why I've been working on it so much hahaha. Thank you so much for the offer, I don't want to be a hassle on you :)

Kind Regards

tatobari, 2016-09-07:
Hi,
I'll see if I cam get the berry today.

Tato

alotred, 2016-09-07:
Hi Tato,
I honestly don't know what to say, you are the most helpful person I have ever talked to. 
I really do not want to be a hassle, so if it is too much trouble please do not worry about it. 
Thank you so much for everything !!!

Kind Regards

tatobari, 2016-09-07:
Hey (still don't know your name, is it Eric?),
Don't worry about it. I'm glad to help. I recieved a lot of help online.
I'm sure the problem is really simple. I'll make a few modifications to the
code so I can make sure there are no overflows.

By the way, an overflow, if you don't know what it is, is when you hit the
limit of the size of the number that can be stored in a certain variable
type. I think some part of the code is hitting that limit but without the
Py it's really hard to debug it.

I'll get back to you in about 12 hours which is when I get home.

Tato

alotred, 2016-09-07:
Hi Tato,
My name is Linda but Eric is close lol
Oh ok that would be amazing!!! I'll be going to bed soon but I should be up by the time you get home.
I will keep looking through and see if anything stands out. 
You are a lifesaver!

Kind Regards

tatobari, 2016-09-07:
Hi!
Sorry about the name. I received a very similar question by some guy called
Eric and thought it was you!

Talk to you later. Good night.

Tato

alotred, 2016-09-07:
Hi, 
hahah that's OK, I just thought you chose a random name lol.

Have a good day at work :)

Kind Regards

tatobari, 2016-09-08:
Hey!
It's 23:00 here and I just got home 20 minutes ago. Long day. I promise I will get the berry tomorrow, make a few tests and then send you another message. I'm really sorry I couldn't make it today.

Tato

alotred, 2016-09-08:
Hi!
That's ok! Thanks for remembering me though:)
Ill go through it in the mean time and see if I figure anything out.

Goodnight!

tatobari, 2016-09-08:
By the way, unless you soldered the "B-" and "B+" pins in your HX711 breakboard, you were not using 32db gain. Only those pins are for 32db gain and they are only for that.

"A+" and "A-" can be used both for 128 and 64. However, if you see more stable values, it's because the resolution is low. According to your logs, there definitely is an overflow which my load cells are not generating and I'm pretty sure I can find it.

tatobari, 2016-09-08:
Just right me and let's see if we can get into a quick call and sort it out. My mail is tato@convoy.me.

tatobari, 2016-09-09:
So, after debugging with "alotred", we found out the order of the bits coming from the HX711 was the problem. My HX711 work in one way and her works in a different way. I'll make the necessary modifications to the code so the order of the bits (and bytes) can be adjusted by configuration.

tatobari, 2016-09-11:
I've made major modifications to the code in order to make it simpler and more _debugable_. The order in which the bytes are used to build the value returned by the HX711 can now be changed using the method `set_reading_format`.

closed 2016-09-12 by tatobari