tatobari/hx711py #6: Wiring the HX711 to a raspberry pi

karl16, 2017-01-13:
I am confused on how to attach from the HX711 to the raspberry pi. In the example code the line
`hx = HX711(5, 6)`
Makes me believe that DAT and CLK are connected to pin 5 and 6 respectfully. But which ones are pin 5 and 6? Because it seems like pin 5 is SCL and 6 is GND which I doubt is correct.

Thanks
Karl

tatobari, 2017-01-14:
Hey Karl,
Dorry for the late response. `5` and `6` refer to the GPIOs 5 and 6 respectively.

[image]

The above layout shows the pin numbers and their names. In this case, the DAT and CLK wires go on pins 29 (`GPIO 5`) and 31 (`GPIO 6`). You can, ofcourse, choose any GPIO's you want, just change the numbers when instantiating the class.

Let me know if I didn't make myself clear or if you have any other doubts.

Regards,
Tato

karl16, 2017-01-15:
Awesome that was super helpful!  Do you also know a good way to connect the load cell to the hx711 because I would rather not solder the wires directly to the hx711

Thanks Karl

tatobari, 2017-01-15:
Hey,
It's good practice to make the connection between the load cell and the
hx711 as direct as possible. I just got Housing and Dupont connectors and
tried to solder everything as tidy as possible.

Pardon me if I'm wrong but my guess is don't want to solder directly to the
hx711 is because you think you might mess it up.

My advice is that you watch any of the many and very good YouTube videos on
how to solder properly, get the correct tools and solder directly. Once you
understand what to do, it's pretty easy.

Some tips:
- use enough solder fluid.
- don't heat the solder tin directly.
- heat the hx711 contact instead.
- use very little solder tin.
- if you mess up, use a solder tim remover and start again.

(Remover:
https://www.google.com.ar/search?q=solder+tin+remove&safe=off&client=ms-android-samsung&prmd=ivn&source=lnms&tbm=isch&sa=X&ved=0ahUKEwj9893I-cLRAhXBDJAKHeLEDkgQ_AUIBygB&biw=360&bih=560#imgrc=_5jqVVpDw6yNIM%3A
)

Regards,
Tato

karl16, 2017-01-15:
Thank you very much! This has been beyond helpful!

Thanks
Karl

closed 2017-01-15 by karl16