tatobari/hx711py #15: remember zero point

sebi85, 2018-05-01:
Hello everybody, I've broken my head for days

I have a Pi3 with HX711

Meanwhile, I've got the balance as a service and loads the values sent via UDP.

Now I have only one problem the balance reset after each restart and that must not be.
Bottles are measured by weight. These bottles can not be taken off the balance every time the machine is restarted but must remain permanently on it.

`#!/usr/bin/env python
import time
import sys
import os
import socket
from hx711 import HX711

def cleanAndExit():
    client.disconnect();
    print "Cleaning..."
    GPIO.cleanup()
    print "Bye!"
    sys.exit()

hx = HX711(27, 17)
hx.set_reading_format("LSB",)

hx.set_reference_unit(92)

hx.reset()
hx.tare()

UDP_IP = "192.168.210.113"
UDP_PORT = 1234

while True:
    try:
        val = hx.get_weight(5)
        print val

   val2 = str(val)
        MESSAGE = "GewichtGas="+ val2
   sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
   sock.sendto(MESSAGE, (UDP_IP, UDP_PORT))

        hx.power_down()
        hx.power_up()
        time.sleep(0.5)
    except (KeyboardInterrupt, SystemExit):
        cleanAndExit()`

tatobari, 2018-05-01:
So, looks like you need your program to remember the `tare`, as far as I understand. I guess the `hx.tare()` method could both, return the tare value so you can store it in persistent memory and support a "manual" value such as `hx.tare(value)`.

Let me know if that's what you're looking for. The code is context agnostic, so it won't persist the value for you if power is lost, that's up to you. However, with this modification, the code would support preset tare value.

Let me know!

sebi85, 2018-05-01:
Yes thats I looking for

tatobari, 2018-05-01:
I've just commited the change. I did not test it cause I don't have my stuff here with me. Let me know if you encounter any issues.

sebi85, 2018-05-01:
thank you so much I test it now

sebi85, 2018-05-01:
German and you?

i will try it I think maybe and then f.read?

val2 = str(val)
	f = open('data.txt', 'w')
	f.write(val2)
	#f.close()

tatobari, 2018-05-01:
Spanish. Sorry, can't help you there, haha.

Saving information in a plain text file is not the most traditional approach but it should get the job done. You should probably do something similar to this:

Define a function to handle it:

```python
def handle_tare():
    offset = None
    f = Path("./data.txt")
    if f.exists():
        try:
            offset = float(f.read())
            hx.set_offset(offset)
        except:
            offset = hx.tare()
            f.write(offset)
```

Then run your code normally, except you'll use `handle_tare()` where you had `hx.tare()`. I didn't test this at all and I simply don't know any Python. Sorry about that.

sebi85, 2018-05-01:
thank you

I did add it in hx711 and change hx.tare() to handle_tare()

but I get this error

`Traceback` (most recent call last):
  File "./skl.py", line 8, in <module>
    from hx711 import HX711
  File "/eseg/hx711.py", line 187
    offset = None
         ^
IndentationError: expected an indented block
``

tatobari, 2018-05-01:
Can't help you from here. I just wanted to point you on the right path. Try checking the indentations but remember that the code I posted needs some work with the file handling.

sebi85, 2018-05-01:
ok that 

have a nice day

tatobari, 2018-05-01:
Same to you and sorry for not being able to help you anymore.

sebi85, 2018-05-01:
every 15 min I get one Gramm lost.

I explain you

i startet the python script and get data 0 after 30 min without change anything is it -2 after some hours -68 .. .. .. ..

tatobari, 2018-05-01:
The same happens to me. I'm still not sure which one is being affected by temperature, the load cell or the HX711. In both cases, resistors play a role in the high sensitivity analog circuits, and resistors are pretty much heat dissipators. My guess is that the load cell is being affected the most.

After a while, it ends up reaching a relatively stable point.

sebi85, 2018-05-01:
i will let it run a few days and look how much it will get dow

tatobari, 2018-05-01:
What's the scale of your load cell?

sebi85, 2018-05-01:
in 1 till 20000 gramm

tatobari, 2018-05-01:
Does is look like this?

[image]

sebi85, 2018-05-01:
it that one https://www.amazon.de/gp/product/B0768CN3T9/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1

tatobari, 2018-05-01:
The error on those cells, which I also have, is in the order of +/- 0,05%. In 20.000 grams, that would be 10 grams. However, that's only the load cell's error. You need to add the HX711 error, which is quite more complicated to analyze and certainly increases the error margin.

If you want 1gr precision in 20kg, you'll need to buy a much much more expensive load cell and Amplified ADC (the green board with the HX711 microchip is an amplified ADC).

sebi85, 2018-05-01:
good to know about. the load cell
but I have the green one (board)

sebi85, 2018-05-02:
i will gone tomorrow snd let you know
thx

sebi85, 2018-05-03:
did you chance the code I can't find hx.tare()

sebi85, 2018-05-03:
now I got it I write to he file but aber restart it bin by 0 again

` import time
import RPi.GPIO as GPIO
import sys
import os
import socket
#import paho.mqtt.client as mqtt
from hx711 import HX711

def cleanAndExit():
    print "Cleaning..."
    GPIO.cleanup()
    print "Bye!"
    sys.exit()

def handle_tare():
    offset = None
    f = Path("data.txt")
    if f.exists():
        try:
            offset = float(f.read())
            hx.set_offset(offset)
        except:
            offset = hx.tare()
            f.write(offset)

hx = HX711(27, 17)
hx.set_reading_format("LSB",)

hx.set_reference_unit(92)

hx.reset()
hx.tare()
hx.set_offset

UDP_IP = "192.168.210.113"
UDP_PORT = 1234

while True:
    try:
        val = hx.get_weight(5)
        print val

	val2 = str(val)
        MESSAGE = "GewichtGas="+ val2
	sock = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
	sock.sendto(MESSAGE, (UDP_IP, UDP_PORT))

	val2 = str(val)
	f = open('data.txt', 'w')
	f.write(val2)
	f.close()

        hx.power_down()
        hx.power_up()
        time.sleep(0.5)
    except (KeyboardInterrupt, SystemExit):
        cleanAndExit() `

tatobari, 2018-05-03:
Hey man, two things.

1. The function "handle_tare()" which I've shared with you needs some work. It's wrong. It won't work.
2. You're not calling the function. According to your code you're doing:

```python
hx.reset()
hx.tare()
hx.set_offset
```
Which should be:
```python
hx.reset()
hx.handle_tare()
```

I can't properly program the function `handle_tare()` right now. Try asking for some help to a friend.

tatobari, 2018-05-03:
I'm closing the issue because it's not related to `hx711py`.

closed 2018-05-03 by tatobari