tatobari/hx711py #18: changing the hx.set_reference_unit() value in example.py doesn't do anything

drapizaaldryx, 2019-02-09:
I'm working with a 3kg load cell and when I run example.py with hx.set_reference_unit(1)
it outputs:
55372
55334
55426
55433
55350
55411
55342
55459
55441
55514
55469
55432
55462
55399

adding 2kg to the load cell changes the output to:
1347015
1347079
1347217
1347085
1347154
1347143
1347165
1347071
1347179
1347080
1347089
1347133
1347182
1346994
1347174
1347168
1347150
1347136
1347186
1347048
1347081
1347139
1347240

i used hx.set_reference_unit(673) since 1347126/2000=673
but the output does not change after running
please help :(

Takrem1, 2019-02-18:
I have the same problem. The values don't change when chaning the reference_unit.

tatobari, 2019-02-18:
A major refactoring was just merged. Could you pull the changes and test again? Thanks. 

Also, please post your test code.

Takrem1, 2019-02-18:
i did

with a reference_unit=1 i got:
-31422
-31384
-31430
-31389
-59393
-31356
-31453
-31389
-31483
-31436
-31376
-31413
-31406
-31436
-31414
-31404

and with reference_unit=92 or any other number:
-31428
-31387
-31373
-31353
-31350
-31430
-31414
-31385
-31417
-31376
-31406

tatobari, 2019-02-18:
@Takrem1 Could you post your code, please? Thanks!

Takrem1, 2019-02-19:
i used the example.py

tatobari, 2019-02-19:
I'll update `README.md` and `example.py` this weekend to make it clearer.

Meanwhile, in `example.py` your getting the raw value from HX711 using `hx.read_long()`. Comment that one and uncomment `hx.get_weight(5)`.

So, change this:
```python
# Prints the weight. Comment if you're debbuging the MSB and LSB issue.
# val = hx.get_weight(5)
val = hx.read_long()
print val
```

To this:
```python
# Prints the weight. Comment if you're debbuging the MSB and LSB issue.
val = hx.get_weight(5)
# val = hx.read_long()
print val
```

rirons, 2019-02-28:
Same issue.  I'm for sure registering changes through the load cell with example.py however these values are still just erroneous long numbers because the set reference unit is having apparently no effect whatsoever up the output?

tatobari, 2019-02-28:
Hey @rirons, are you using `example.py`? Did you modify any lines?

rirons, 2019-02-28:
Using example.py.  Other than the reference unit everything is as downloaded.

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

tatobari, 2019-02-28:
Please, read my comment just above your first one on this thread.

closed 2019-03-06 by drapizaaldryx