tatobari/hx711py #21: Reference unit and settings

cbkn76, 2019-03-23:
Hello, and I appreciate any help I can get.
I am using 4  degraw 50kg sensors. Using  Standard example.py, I have only changed the reference unit, and commented read long and uncommented get weight. Also tried playing with uncommenting hx.tare(). 

As I’m completely new to python I am pretty sure I am doing something wrong and need clarification.

1. Should hx.tare be commented or uncommented, or changed after calibrating? I only ask because comments on SparkFun page said they had to uncomment to get it to work.

2. Should calibration be done using read long or get weight?

Using get weight I get readings of around 32,000 with no weight, and that does not change when I add weight. Changing reference unit can bring that number down, but will still not read weight.

Again I appreciate any help or clarification for my troubleshooting. Been trying to figure this out for two days but some information on other sites has confused me.

tatobari, 2019-03-23:
Hey! Please post a picture of your wiring or your curret schematics. Also, could you share the Sparkfun link where you read about this? Finally, which HX711 breakout board are you using? A good picture would be useful.

cbkn76, 2019-03-23:
Sorry, wasnt sparkfun. here is the link
https://tutorials-raspberrypi.com/digital-raspberry-pi-scale-weight-sensor-hx711/

schematic i used is this
[image]

trying to get pictures up. i believe i saw that you use the same hx711 in another post. i have soldered it

cbkn76, 2019-03-23:
[image]
[image]

cbkn76, 2019-03-23:
[image]

cbkn76, 2019-03-23:
[image]

tatobari, 2019-03-25:
Please, post a picture of the wires on the Raspberry.

By the way, `hx.tare()` is a function that should be used to tare the scale you've built. The function will produce an offset value which is `self.OFFSET`. This means that the scale shouldn't have any weight until that function has run.

`example.py` is a very dumb concept. You should have a push button or something else that let's you run `hx.tare()` when needed.

cbkn76, 2019-03-25:
[image]

tatobari, 2019-03-25:
Looks good. Could you post the values you're getting?

cbkn76, 2019-03-25:
Best picture I have as I am traveling at the moment, I appreciate the help. So do I get the reference unit when using get weight or read long?
This troubleshooting is driving me nuts.

tatobari, 2019-03-25:
- Leave `hx.tare()` uncommented.
- Always wait until `hx.tare()` has run before adding weight.
- The first time, make sure that the reference unit is `1` by using `hx.set_reference_unit(1)`.
- To get the new reference unit, run run `example.py` with `val = hx.get_weight()` uncommented and comment `val = hx.read_long()`, wait for `hx.tare()` and then add something of which you know its weight. Preferably, something that weights 1kg.
- Take not of what `hx.get_weight()` returns. That value is now your reference unit because it's what 1kg means for your current layout.
- Now use that value for `hx.set_reference_unit(VALUE)`.

cbkn76, 2019-03-25:
I’ll have to get back to you on Wednesday, I have tried so many combinations my mind is jumbled. The main problem is the readings not changing when I apply weight. Or not adjusting when I change the reference unit.

It is possible one or more of the sensors is bad, so I think I am going to order another kit. Only reason I think one may be bad is because it doesn’t seem like this should be that hard.

tatobari, 2019-03-25:
I'd probably check the sensors too. Good luck.

slabula, 2019-04-27:
Try making a wheatstone bridge:
https://en.wikipedia.org/wiki/Wheatstone_bridge
And connecting that to your Raspberry Pi/HX711 board.  That you can verify that your boards work.

tatobari, 2019-05-19:
Closing this issue as it isn't due a problem with the library itself.

closed 2019-05-19 by tatobari