tatobari/hx711py #22: Multiple instances gives AttributeError

penbob3, 2019-05-07:
I want to read the values from 3 separate load cells, each with its own HX711. I need the weight value to stay the same from each reading, or basically, I need to read the values from each load cell without tareing them. Is there a way to do this without having multiple HX711 instances?

(Example code only shows two instances)
`from hx711 import HX711
hx1 = HX711(5, 6)
hx1.set_reading_format("MSB","MSB")
hx1.set_reference_unit(687)
hx1.tare()
hx1.power_down()
hx2 = HX711(7, 8)
hx2.set_reading_format("MSB","MSB")
hx2.set_reference_unit(687)
hx2.tare()
hx2.power_down()

if selectedlc == 1:
    hx1.power_up()
    intWeight = hx1.get_weight(5)
    hx1.power_down()
if selectedlc == 2:
    hx2.power_up()
    intWeight = hx2.get_weight(5)
    hx2.power_down()`

tatobari, 2019-05-19:
Could you please post the error you're getting? Thanks!

MrBudo, 2020-06-16:
[image]

I am getting this attribute error.
Also I am getting super random values when I uncomment the 3 line it throws the error

NeonSpork, 2020-06-16:
Those lines are commented out in the example, and when I look at the current hx711.py file I can't find those functions defined anywhere. Maybe that's a hold over from a older version? @tatobari maybe you remember? Comment those out or delete them and the AttributeError will go away at least.

MrBudo, 2020-06-16:
[image]

I am getting super random values. I have tried changing LSB and MSB. changed it to 3.3 V still no luck. I really trying hard to find a solution. Can you suggest me something that I can try.

I am using this sensor with hx711

MrBudo, 2020-06-16:
https://www.ebay.com.au/itm/1x-Miniature-Load-Cell-Weighing-Sensor-Pull-Rod-Pressure-Tension-Sensor-0-100kg/293280015699?hash=item4448d9e553:m:m5ou6L29ZJbBRaHQp9x_SEw

NeonSpork, 2020-06-16:
Wait hang on, is this a separate issue than the one in the first post?

Regarding the randomness of the readings maybe try setting a lower gain and recalibrating.

tatobari, 2024-03-07:
@penbob3 Are you still interested on using this library and solving this issue?