tatobari/hx711py #23: Giving Random Values

kush-13, 2019-06-11:
My Hx711 and Load Cell works fine with Arduino Uno  (giving outputs with +-4 grams of fluctuations ).
But when I run example.py  on my Rpi with following connection:
DT>>5
SCK>>6
VCC>>5V
GND>>GND(RPi 3B+)

I'm getting random values even when nothing placed on load cell.

I get the following result when I run  example.py with nothing placed on the Loadcell:

***************************************************
-3939697
-4791665
123791
-1186929
-859249
-4595057
-1252465
-3546225
-7150961
-3808625
-7019889
-4529521
-7478641
-8854897
-4005233
-7609713
-7347569
-9706865
-6888817
-6692209
3859087
-73073
3138191
320143

***************************************************

Plz Help !!!

I'm using python2.7 to run example.py which comes pre installed on Raspbian which is running on my RPi 3B Plus

tatobari, 2019-06-11:
When did you start using the library? What's the current commit you've got
cloned?

kush-13, 2019-07-06:
I UnderVoltaged  the sensor{ HX711 24 Bit ADC } to 3V3 and then it started giving me stable values .

closed 2019-07-06 by kush-13

tatobari, 2019-07-06:
I use it with 5V and it works fine. What voltage where you using it with?