tatobari/hx711py #24: Wiring dual channel A + B

hyptechdev2015, 2019-06-17:
Hi,

Will this wiring work for both channel from @kozoke
#20 

[https://cloud.githubusercontent.com/assets/14950116/15843880/61435e10-2c1a-11e6-87b2-18cf1d95ee6b.jpg](url)

reference from: [https://github.com/bogde/HX711/issues/12#issuecomment-224145762](url)

tatobari, 2019-06-18:
Hi! Your question references data from other projects and it's more related to how to combine 3 wire load cells with two different gains, which is pretty much out of the focus of this project.

In your case, I'd check introductions to load cells such as [Sparkfun's one](https://learn.sparkfun.com/tutorials/load-cell-amplifier-hx711-breakout-hookup-guide/all), which is very good and it's also the one I used when I started working with load cells.

closed 2019-06-18 by tatobari

kozoke, 2019-06-18:
> Hi,
> 
> Will this wiring work for both channel from @kozoke
> 
> #20
> 
> [Picture Link](https://cloud.githubusercontent.com/assets/14950116/15843880/61435e10-2c1a-11e6-87b2-18cf1d95ee6b.jpg)
> 
> Referenced from: [This comment on Issue 12](https://github.com/bogde/HX711/issues/12#issuecomment-224145762)

I went to two separate Hx-711s and actually two esp boards.  All separate and all working.  I also now use esphome on them (just usuign 1 hx711) and it works great!  Highly recommend esphome if you are looking for a push-button simple set-up.  https://esphome.io/  (I have not tried configuring it for two load cells - but I shoud some day).

hyptechdev2015, 2019-06-18:
thank you @tatobari  and @kozoke 

I ended up using Channel A only, and it works for me with your codes @tatobari