tatobari/hx711py #29: example.py comments

koenner, 2019-09-08:
I am new to coding and really appreciate you sharing this content.  Looking at "example.py"; would it be possible to comment what each of the lines does?  For example, trying to understand what "EMULATE_HX711=FALSE" does..

NeonSpork, 2020-03-05:
EMULATE_HX711 is a True/False flag that the user sets at the beginning of the script. If it is True then the hx711 module is imported from a module that emulated the hx711, aka the result is simulated. If EMULATE_HX711=False, as is the case in the example.py script, then the program imports the _real_ hx711 module and will only work if it is run on hardware that has the RPi.GPIO library and HX711 library installed (most likely a Raspberry Pi) and has a HX711 properly connected via the GPIO pins.

tatobari, 2024-03-07:
@koenner Are you still interested on using this library and solving this issue?