tatobari/hx711py #32: Having problem on assigning variables

kumucho, 2020-03-04:
Hi mr.tatobari,
forgive my ignorance
I am having trouble on assigning variable to get an initial data using my 20kg Load cell bar
and HX711
I copied one of your lines and made a test run.
I got a message of syntax error. Is the "val" the only variable i can use?
I am trying to get a percentage ratio from the initial weight and the current weight but still do not
have an answer for almost 3weeks.

This here is one of the lines i am using in my RPi3
  try:
        val = max(0, int(hx.get_weight(5)))
        mylcd.lcd_display_string("Load={0:0.1f}g" .format(val),2)
        hx.power_down()
        hx.power_up()
        print(val)
	val2 = hx.get_weight(5)
	mc = (1 - (val2-val)/val2) * 100
	mylcd.lcd_display_string("MC={0:0.1f}%" .format(mc),4)
        
        
    except (KeyboardInterrupt, SystemExit):
        cleanAndExit()

P.S your codes and lines are being used in education not on any kind to get an income
P.P.S specifically, on my thesis project

I am hoping and wishing to get a response. Big thanks for even noticing me.

NeonSpork, 2020-03-04:
Would you mind posting the syntax error and what it says specifically? Syntax error is an error in the python code and could be a multitude of things.

kumucho, 2020-03-04:
> Would you mind posting the syntax error and what it says specifically? Syntax error is an error in the python code and could be a multitude of things.

Thanks for the reply Mr.@NeonSpork,

The exact scenario is after I run the program, a message box pops up on Python IDLE and says a syntax error then highlights my variable "val2" or any variable i had tried.

I cannot send some screenshot as of this moment due to my prototype is being constructed. Ill be sending it some later time.

NeonSpork, 2020-03-04:
Are you sure you have the indentation correct? The code immediately after "try:" must all be indented the same way as after "except:" in your example. Does it look like this?
```
try:
    val = max(0, int(hx.get_weight(5)))
    mylcd.lcd_display_string("Load={0:0.1f}g" .format(val),2)
    hx.power_down()
    hx.power_up()
    print(val)
    val2 = hx.get_weight(5)
    mc = (1 - (val2-val)/val2) * 100
    mylcd.lcd_display_string("MC={0:0.1f}%" .format(mc),4)

except (KeyboardInterrupt, SystemExit):
    cleanAndExit()
```

Either way I think the specific syntax error would be helpful if you can post that when you get a chance. :)

kumucho, 2020-03-04:
> Are you sure you have the indentation correct? The code immediately after "try:" must all be indented the same way as after "except:" in your example. Does it look like this?
>
> ```
> try:
>     val = [...]

Yes. I have indented it correctly. Sorry about my first post, i didn't change it when github removed the indents.

NeonSpork, 2020-03-04:
OK good, just had to check. Is the rest of your code the same as the code in example.py? Unfortunately it's going to be hard to help you further without more details specific to the syntax error to be honest.

kumucho, 2020-03-04:
> OK good, just had to check. Is the rest of your code the same as the code in example.py? Unfortunately it's going to be hard to help you further without more details specific to the syntax error to be honest.

From first line until while True line.
I changed the val variable and added max 0 to 5(dunno what happened with these codes i added)

I also set the referrenceUnit to -121 cause it shows me much closer value on grams of 3kg.

Then added a code to print it on LCD Display. 

My main goal is to create a variable so that I can present an output on LCD Display using the first weight it recorded then the weight as time goes by. A percentage ratio between two values.

NeonSpork, 2020-03-04:
In the picture you posted I can immediately see that you're missing two )) at the end of the line above. It should be:
```
val = max(0, int(hx.get_weight(5)))
```
The last two brackets are missing somehow :) Hopefully that solves your problem!

kumucho, 2020-03-04:
> In the picture you posted I can immediately see that you're missing two closing brackets like this **)** at the end of the line above. It should be:
> 
> ```
> val = max(0, int(hx.get_weight(5)))
> ```
> 
> The last two brackets are missing somehow :) Hopefully that solves your problem!

Wow. Im really embarassed to waste your time. Thank you so much. It is the problem, i have new problem at hand.

Thankyou sir @NeonSpork !

NeonSpork, 2020-03-04:
Hey, no problem! Sometimes all it takes is a fresh pair of eyes! Happy to help :)

closed 2020-03-05 by kumucho