tatobari/hx711py #33: Getting the first weight value aside from tare value

kumucho, 2020-03-05:
Hi sir tatobari,

Do you have some way to get the first initial mass from the code?
like.. after the Tare where the value is 0, then you suddenly lumped in a 3kg you get to record and save the 3kg value sensed by the loadcell to be used in equation like these:

`MC = (1-(initial mass - present mass)/initial mass) * 100`

Ive tried this code and got myself a 1% because of lack of initial mass value:
[image]

NeonSpork, 2020-03-05:
So if I understand correctly, you're trying to first record a weight, then record a _new_ weight, followed by calculating the % difference in weight. Correct?

If that's true then your method is likely not working because both val and val2 are in the same while True loop, and they value is being set essentially simultaneously. So if val = val2 your equation would be like the following:
`MC = (1 - 0) * 100 = 100`
However in the code you took a screen shot of I can see that your equation is different and you don't have brackets around the 1 - ((val2-val)/val2) * 100. This means that your ((val2-val)/val2) value is multiplied by 100 **before** it is subtracted from 1, aka 0 * 100 = 0. Which will always give you MC = 1 in your case, hence your reported value of 1%.

First of all you'd need the proper brackets, but if you want to compare values before and after I think you're going to have to modify the code to store a value **outside** the while True loop (which is run as fast as it can by the python script as long as it's running), and use that in stead of val. Or move val out of the while True loop somehow.

Does that make sense? Hope I understood what you want to achieve correctly. Good luck!

kumucho, 2020-03-05:
Thankyou for your reply sir @NeonSpork.
Yes, I am trying to record the first weight then use the present weight to achieve the percent ratio. Therefore, the formula will be (initial mass - present mass)/initial mass.

Ohh, right I really didnt put a bracket on the formula. Thats why I am getting a 1% output.

And one more question,
What condition should I use if I'll set the val aside? I am thinking of using try but it bothers my mind that it may only get the tare weight because of it having to get or work on a moment of time.

Can I use "try" to get the value of weight ive put after getting the tare within .....lessay 5secs?

Once again,
Thankyou for your help. You are a lifesaver

NeonSpork, 2020-03-05:
Well I'm not sure what use case you have for this. Is this something that _needs_ to be on a loop and run automatically? Or is this perhaps something you can run every time you want to check the change? Your use case will probably determine the best way to implement what you want to be honest.

closed 2024-03-07 by tatobari