tatobari/hx711py #35: No tare

DavidMolevelt, 2020-06-03:
Dear,
I am looking for a change in the program. Once the scale is set, tare is known, it should run at startup without rescale. Is it possible to change this?

girardcarillo, 2021-03-31:
Hi,
I think I have kind of the same question.
I would tare only at first use, and then be able to run again the code without taring again.
I suppose we can edit the emulated_hx711.py ?
Thanks a lot :)

girardcarillo, 2021-03-31:
What I have found:
go to hx711.py and fine the method tare_A()
After the line `hx711.py` add  `print(value)`
Save the file and with nothing on your scale run the example.py code.
The printed value is what you need to save somewhere if you don't want to tare again.

I'm not saying this is a solution to our issue, but a way to find the solution.

If you have any suggestion how to proceed, I would be glad to hear it :)

GC