tatobari/hx711py #37: Is it possible to connect multiple sensors to a single pi with this code?

ronanb95, 2020-10-19:
Hi,

The code works great for a single hx711 connected to one pi. 

I am trying to connect multiple load sensors to a single pi and I am just wondering is it possible to do this by changing part of this code?

I have tried creating a new instance like hx711(8,9) where 8 and 9 are the corresponding pins connecting to the hx711 but this does not seem to be working. I am very new to all of this so I could be doing it completely wrong.

Ideally I would like to be able to connect 4+ sensors but even if I could connect 2 simultaneously then that would be great.

Thanks

Janik-97, 2020-11-12:
Until now I have connected 4 hx771 to one RPi 3 B+ and they work perfectly.
Just use different pins and wire them correctly (pinout.xyz).
My goal is to run 12 hx711 simultaneously. Should be no problem in theory.

ronanb95, 2020-11-12:
Hey Janik,

I have since connected another unit with no problems at all. 

As far as I can tell the maximum number of units that can be connected at any one time would be 4 because each unit must be connected to one of the 4 power pins on the pi. I could be wrong with that theory though.

Thanks for the reply anyway :)

NeonSpork, 2020-11-13:
I'm fairly sure you could piggyback some of the power from the pins with
some creative wiring to run even more in that case, I doubt the hx7711 need
an immense amount of juice. :)

NeonSpork, 2020-11-21:
I guess this issue can be closed since the question has been answered.

closed 2020-11-30 by ronanb95