tatobari/hx711py #38: I am getting zero values (all ways) continuously irrespective of the load

DATTUSAM, 2020-10-24:
Hello, I am getting zero values so I followed the comment

I am using GPIO pin5 & 6 (GPIO5 for DT & GPIO 6 CLK)

Tried changing power source to both with VCC 5V and 3.3V

Tried with Different set_reference_units

still I getting zero values.

Please help me I am really determined to do it.

aggounix, 2021-01-22:
Hello, did you fix this problem, I am getting the same one

tatobari, 2021-02-25:
I've tested the library and it seems to be working just fine.

It's most important that you submit a schema of how you've wired the loadcells to the HX711. The 50kg loadcells everyone uses, the square ones, don't have an embedded wheatstone bridge and therefore must be connected in a wheatstone bridge layout. Posting your schema will help me help you debug your issue.

monacles, 2021-02-28:
I'm experiencing the same issue. I just get zeros printed continuously. Here is the load cell setup:[image]

Greenbauer, 2021-08-18:
had the same issue. changing the mode from BCM to BOARD fixed it for me.

KaitoCross, 2022-08-13:
I have the same issue, but I'm using a 20kg load cell (beam type). CLK on Pin 20, DT on 21. Chip & load cell are working on my Arduino, but is giving me zeros on the Raspberry