tatobari/hx711py #41: 'HX711' object has no attribute 'get_weight'

Schmediii, 2021-04-17:
Hello everyone,

I tested your programm, but i have a error: AttributeError: 'HX711' object has no attribute 'get_weight'.

What have i done wrong? Can someone help me.

Thank you

tatobari, 2021-04-17:
Any chance you couls post your code so we know how you've implemented it?

Schmediii, 2021-04-17:
Yes shure, the code is from the example.py:

`#! /usr/bin/python2

import time
import sys

EMULATE_HX711=False

referenceUnit = 1

if not EMULATE_HX711:
    import RPi.GPIO as GPIO
    from hx711 import HX711
else:
    from emulated_hx711 import HX711

def cleanAndExit():
    print("Cleaning...")

    if not EMULATE_HX711:
        GPIO.cleanup()
        
    print("Bye!")
    sys.exit()

hx = HX711(5, 6)

# I've found out that, for some reason, the order of the bytes is not always the same between versions of python, numpy and the hx711 itself.
# Still need to figure out why does it change.
# If you're experiencing super random values, change these values to MSB or LSB until to get more stable values.
# There is some code below to debug and log the order of the bits and the bytes.
# The first parameter is the order in which the bytes are used to build the "long" value.
# The second paramter is the order of the bits inside each byte.
# According to the HX711 Datasheet, the second parameter is MSB so you shouldn't need to modify it.
hx.set_reading_format("MSB", "MSB")

# HOW TO CALCULATE THE REFFERENCE UNIT
# To set the reference unit to 1. Put 1kg on your sensor or anything you have and know exactly how much it weights.
# In this case, 92 is 1 gram because, with 1 as a reference unit I got numbers near 0 without any weight
# and I got numbers around 184000 when I added 2kg. So, according to the rule of thirds:
# If 2000 grams is 184000 then 1000 grams is 184000 / 2000 = 92.
#hx.set_reference_unit(113)
hx.set_reference_unit(referenceUnit)

hx.reset()

hx.tare()

print("Tare done! Add weight now...")

# to use both channels, you'll need to tare them both
#hx.tare_A()
#hx.tare_B()

while True:
    try:
        # These three lines are usefull to debug wether to use MSB or LSB in the reading formats
        # for the first parameter of "hx.set_reading_format("LSB", "MSB")".
        # Comment the two lines "val = hx.get_weight(5)" and "print val" and uncomment these three lines to see what it prints.
        
        # np_arr8_string = hx.get_np_arr8_string()
        # binary_string = hx.get_binary_string()
        # print binary_string + " " + np_arr8_string
        
        # Prints the weight. Comment if you're debbuging the MSB and LSB issue.
        val = hx.get_weight(5)
        print(val)

        # To get weight from both channels (if you have load cells hooked up 
        # to both channel A and B), do something like this
        #val_A = hx.get_weight_A(5)
        #val_B = hx.get_weight_B(5)
        #print "A: %s  B: %s" % ( val_A, val_B )

        hx.power_down()
        hx.power_up()
        time.sleep(0.1)

    except (KeyboardInterrupt, SystemExit):
        cleanAndExit()`

And this is the error message:

`Tare done! Add weight now...
Traceback (most recent call last):
  File "gewicht1.py", line 65, in <module>
    val = hx.get_weight(5)
AttributeError: 'HX711' object has no attribute 'get_weight'
`

carlosbarilatti, 2021-04-17:
Hummm, this repository doesn't have any file called gewicht1.py. Is there a chance that you might be calling hx.get_weight() from a file.on your code called gewicht1.py but you didn't import hx711 correctly?

Schmediii, 2021-04-17:
Well, i made a repository for th gewicht1.py.
Bevore i installed hx711 with the code: pip3 install HX711

I now tried it with this tutorial: [https://tutorials-raspberrypi.de/raspberry-pi-waage-bauen-gewichtssensor-hx711/](url)

So i installed the library with: git clone https://github.com/tatobari/hx711py

Then in the repository: cd hx711py

I run then the example.py

Now the code works but i become just many 0.0. Is this right?