tatobari/hx711py #50: Random Values on the raspberry pi

programmeddeath1, 2023-01-07:
Im trying to implement this code with the Raspberry pi 4b, on python3. It is giving random values, I tried switching between MSB/LSB, I tried all combinations of the load cell wires to check if there is a difference between the supported formats of hx711 and the load cell that I have. I am not able to debug from the code because the binary values array function doesnt exist. Can you help me understand what mistakes I might be making.
[image]
[image]
[image]
[image]

From the sensor I have taken the red and black to E+ and E-, The green and white as A+ and A-.

Please help me im stuck on this for quite a while I need to figure out a solution for delivery as soon as possible. Any help would be appreciated!

Morrious, 2023-01-07:
This isn't my repo, but you should first check all your wiring matches the GPIO pins you have set. You should have the clock/sck pin and GPIO input pin wired to the HX711, and then the 5v power and ground. Make sure you've configured your GPIO properly as well.

Over on my similar repo, I have a raw GPIO test you can use to see exactly how the GPIO is interacting with the hx711.

programmeddeath1, 2023-01-07:
Hey yes I have assigned data and sck to gpio 5 nd 6 as mentioned in the repo demo. I have checked the GPIO configuration is set as per the two tutorials mentioning this repu, not the connections towards the sensor side I am not sure if the pin connections to the given sensors and hx711 are proper. The sensor has 4 cables of red,white,black and green. I have attached the red and black to excitiation(E+,E-) and green and white to A+ and A-(tried both) I am still getting random values from the hx711. I tried with another weight sensor and still the resut is the same. Is it a fault with the code, is it sending data differentl to the rpi than what the code in the repo ismeant to handle? Ill ttest the raw GPIO from your code once.

Morrious, 2023-01-07:
Here's a wiring diagram I have of the HX711s I use.
[image]