tatobari/hx711py #58: Potential danger wiring HX711 to 5V w Raspberry Pi

ASebastian27, 2024-05-22:
Hi,

I believe it is not ideal to indicate that the HX711 should be connected to 5V. That is totally fine with an Arduino, but probably unsafe with the RPi.
I am not sure what exactly might happen, but the RPi (especially older ones, I guess?) might get damaged in the process.

I will be linking this conversation: https://forums.raspberrypi.com/viewtopic.php?t=335497.

_"Arduino UNO's GPIOs maximum current is rated at 40mA at 5V, whereas RPi's is 3.3V at 16mA."_
Thus, making the HX711 run at 5V with the RPi might not be safe.

Unless... am I missing something?

tatobari, 2024-06-11:
Hi! Thank you for raising this issue. 

I'm not sure if anywhere on this repository it's indicated that the HX711 should be powered by the 5V pin. Could you point me to where you found that?

Anyway, if I'm not wrong, the 5V pin is directly wired to the power source powering the Raspberry Pi, so the current limit is set by the power source.

Jai-GAY, 2024-11-05:
PRO TIP: In many cases, you can just short VCC and VDD together. If your microcontroller uses 3.3V logic however, you'll want to connect [VCC to 5V and VDD to 3.3V](https://learn.sparkfun.com/tutorials/load-cell-amplifier-hx711-breakout-hookup-guide).